sanny32/OpenModScan #2: Offset +1 on all registers

MarkoBursic, 2023-07-07:
If I read holding register  Address 0001 it actually tries to read register 0, so exception is throwed. To read register 1 you have to set Address to  0002. It applies to all kind of registers.

[image]

[image]

sanny32, 2023-07-07:
Please let me know from what register 0 or 1 address space started in your device.

MarkoBursic, 2023-07-07:
After examination of the device (Corrigo HVAC controller) and reading the Modbus protocol specs I came to conclusion that maybe the device protocol is implemented in incorrect way. For example, from manual the outdoor temperature is input register 1, translated in modbus space this shall be input register 30001, then subtract 30001 as per modbus specs, the query address shall be 0000 (blue numbers in Wireshark debug), but it throws error, this for any range of registers (coils, contacts, input registers, holding registers).

Short: Closed, Modbus device incorrectly implemented.

closed 2023-07-07 by MarkoBursic