sanny32/OpenModScan #6: How to install?

hagenbuch, 2024-01-03:
Hi all,

I would like to install the software on a Linut Mint (22?) with Cinnamon desktop.

I installed the package(s) (amd64 version) and I'd expect that the installation gives a hint if Qt5 or Qt6 is present or needed (I have no idea what that is, I'm more like an engineer).

Both packets get installed with the info "already installed". Then I go to /opt/OpenModScan and run omodscan.sh only to see this:

    qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "/opt/OpenModScan/plugins//platforms:" even though it was found.
    This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

    Available platform plugins are: minimalegl, wayland, offscreen, eglfs, vkkhrdisplay, linuxfb, wayland-egl, minimal, vnc, xcb.

Sorry I have no idea what to do next and for I might not be the only one I mustered up the courage to post here :-) in hope for a hint.

sanny32, 2024-01-03:
Hello @hagenbuch. It seems like some libraries are missing. Can you give me a full linux version, i’ll test application.

hagenbuch, 2024-01-03:
Sure! Wow that was fast, many thanks!

    cat /etc/lsb-release 
    DISTRIB_ID=LinuxMint
    DISTRIB_RELEASE=21.2
    DISTRIB_CODENAME=victoria
    DISTRIB_DESCRIPTION="Linux Mint 21.2 Victoria"
    uname -a
    Linux <machine> 5.15.0-91-generic #101-Ubuntu SMP Tue Nov 14 13:30:08 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

Hope this is the right thing..

sanny32, 2024-01-03:
You must install libxcb-cursor0 library via
`sudo apt install libxcb-cursor0`

hagenbuch, 2024-01-03:
That was it! Fantastic, many thanks! May it help others, too!