sanny32/OpenModScan #9: address

tosione, 2024-06-17:
There seems to be an offset of 2 in the address, at least with the 32bit data with Modbus TCP (I can only test the Modbus on a machine that only uses 32bit data: long int, or float32). 

The get the correct data I must increase the address by 2, as it can be seen on screenshot:

Tested on version 1.7.0.

[image]

sanny32, 2024-06-18:
Hello @tosione. Why do you think there is a offset in addresses here?

tosione, 2024-06-18:
Because I know the values to expect at each address, as we design the machine: values of ~411 are expected on addresses 16078, 16080, 16082, not the ones before or after.
Besides, with other programs (QModMaster and Radzio! Modbus Master Simulator ) this coincides.

sanny32, 2024-06-19:
@tosione can you show hex values of your server address space?

tosione, 2024-06-27:
Sorry for delay @sanny32,

It is a bit difficult, and possibly senseless, to show you my hex values from server address (microcontroller): that's because we are using a Raspberry Pi as a gateway, and the software on the Raspberry is not easily available to me:

_Windows-PC ---(ModBus TCP)---> Raspberry-Pi  ---(ModBus RTU)---> Microcontroller_

I can't assure that the gateway is not adding any address offset, but again, this do not happen with other programs. 

Reading on the internet I have seen that address offset is not such a uncommon problem, and many programs offer a way the configure this offset (at least for configuring the 0 or 1 based addressing). Maybe it would be possible to implement this, to allow greater compatibility.

sanny32, 2024-06-28:
Yes, the base address in Open Modscan is 1, not 0. If your device has base address 0 you can think, that the program has address offset.

sanny32, 2024-10-16:
Hello @dangkhoa1612. This may be possible to do in one of the future releases.

sanny32, 2024-10-24:
Implemented at version 1.8.0

closed 2024-10-24 by sanny32