sanny32/OpenModScan #15: Connection Issue

s123sahir, 2024-10-09:
When trying to connect, i am getting connection error. I tried different ports as well. Also tired after turning off the firewall.

[image]

[image]

sanny32, 2024-10-09:
Hello @s123sahir. Tcp socket error occurs when connection via tcp protocol faild. You must check your network settings. 
Can you check ip address availability via ping or telnet?

lexstarwind, 2024-11-22:
I also have a strange TCP connection issue with this app. I'm testing the number of concurrent TCP connections on a device I have. The device supports up to 4 connections and will reject the 5th. I've testing using Open ModScan and ModScan64. I open 5 instances of Open ModScan and the first 4 can connect with no problem. When I try to connect the 5th instance, the connection is rejected as expected. After that happens, the other 4 instances are disconnected and can no longer connect to the device until the device is rebooted. Not sure if it's an issue with my device or with Open Modscan. The strange thing is that after all instances of Open Modscan disconnect, they can no longer connect until I reboot the device. However, modscan 64 can still connect and read data when this happens. I'm not 100% sure where the problem lies but I thought I would mention it. It's also worth mentioning that Modscan64 itself does not cause the system to enter this weird state. I can connect 4 instances of Modscan64, the 5th connection is rejected and none of the other connected instances of Modscan64 seem to be affected.

sanny32, 2025-02-03:
Hello @lexstarwind  
It seems that the problem here is not in the openmodscan program

sanny32, 2025-12-04:
Hello @msaebani-cmyk. You must setup correct device id.