sanny32/OpenModScan #18: Ubuntu 24.04 installed application do not have rights to acces tty*

talenat-st, 2024-11-03:
Installed deb do not have access to tty*
I think something should be done with that as only chmod 666 tty* helps but until reboot.
Deb package shoul deal with that.

talenat-st, 2024-11-11:
Hi,
It is not a point of doing mysef when installation program should do this.
I am aware of adding user to dialout group and as I already wrote I can chmod tty to use it while I need it but that is also not a point.
My comment was in order to improve great progam even more.

ortegahernandes, 2024-11-11:
The comment was just to avoid every time you restart the operating system you don't need to release the port. 

with dialout it becomes permanent, that is, it can be restarted.

 I agree with you that when installed, it would be good to have the serial port already released, especially for those who are not very familiar with Linux, which is why I also suggested dialout.

sanny32, 2024-12-05:
> It is not a point of doing mysef when installation program should do this.
This not possible to do via deb package

talenat-st, 2024-12-05:
Hi sanny32,
I can not tell you how you can do it but I am sure deb package can do it as my SCADA package is installed as deb package and installation makes changes so scada service can access same tty without any issue.
After installation my SCADA can work with tty in ModbusRTU driver without my intervention.
Best regards

sanny32, 2024-12-05:
The problem is that the deb package is installed by the root user and there is no way to know which user to add to the 'dialout' group.
In your case, the service is probably running as the root user and has access to serial ports.

talenat-st, 2024-12-05:
Yes that is true. Service is running as root user.
Maybe command line switch with username could be option for install so that install script know user that needs to put into dialout group?

ortegahernandes, 2024-12-14:
I don't know if it's possible. 
But it would be at the time of installation or to open a typing box for the user to type their username. 

Another alternative is a configuration button when the user is no longer in root mode. 

If it's not possible, at least a warning or a tip that if the serial doesn't work, try to run the command.

sanny32, 2024-12-16:
I added access to /dev/ttyS* devices via udev rules. Can you test these installers?
Please, unzip files before run deb installation.

[qt5-omodscan_1.8.2-2_amd64.deb.zip](https://github.com/user-attachments/files/18149364/qt5-omodscan_1.8.2-2_amd64.deb.zip)
[qt6-omodscan_1.8.2-2_amd64.deb.zip](https://github.com/user-attachments/files/18149350/qt6-omodscan_1.8.2-2_amd64.deb.zip)

ortegahernandes, 2024-12-17:
In my case it's a ttyusb, I don't know if I've entered this case, but I tried here anyway. 

I did a clean install of Ubuntu in VirtualBox and tried to run it directly. 

It only worked when I added the user to the dialout, when I removed it it stopped again. 

In this port error warning, there could already be a hint on how to solve the problem 😉

[image]
[image]

Oh, one observation, in Ubuntu it works normally, in Linux Mint the software seems to have a library problem, I didn't want to go into too much detail.

sanny32, 2024-12-17:
Thank you for testing. I will add access to /dev/ttyUSB* to udev rules to.

talenat-st, 2024-12-18:
I can confirm that it is working with real serial port /dev/ttyS0.
Adding /dev/ttyUSB* will solve USB to serial adapters and that would be complete fix.
Great

sanny32, 2025-02-27:
Implemented in version 1.9.0 deb packages.

closed 2025-02-27 by sanny32