sanny32/OpenModScan #31: mismatched holding and input registers

agorsak, 2025-02-07:
in the scan interface dropdown selection MODBUS point type the holding registers and input registers are prefaced with wrong number:
03: HOLDING REGISTER should be 04: HOLDING REGISTER and 
04: INPUT REGISTER should be 03: INPUT REGISTER

The adresses below are correct.

[image]

sanny32, 2025-02-08:
Hello @agorsak. 03 and 04 is the modbus function numbers, not a 3x and 4x.

Afroboltski, 2025-03-18:
This is just a weird quirk within Modbus. I'm assuming someone made an error many years ago, which will persist through the ages.

Holding regs are 4xxxx and read using F/C 3
Input regs are 3xxxx and read using F/C 4

e.g. [https://www.fernhillsoftware.com/help/drivers/modbus/data-address-format.html#:~:text=The%2016%2Dbit%20number%20gives,the%20second%2C%20and%20so%20on.&text=The%20traditional%20Modicon%205%20digit,first%209999%20coils%20or%20registers.](url)

closed 2025-06-09 by sanny32