sanny32/OpenModScan #33: Crashes when changing datatypes

daxxx, 2025-03-05:
When in 'Show data' screen and clicking on another datatype in 90% of all cases the program just exits without giving an error.
It gets more likely when length gets greater.

OS: Windows 11

sanny32, 2025-03-05:
To reproduce bug need more information. Please give me steps to reproduce bug

daxxx, 2025-03-05:
Connect to a modbus.
Open Show data.
Set length to greater than 100.
Change data types.

sanny32, 2025-03-05:
What version do you use?
Are you connecting to modbus rtu or modbus tcp?
Which data do you receive from modbus server (screenshot)?
May be can you add video?

Afroboltski, 2025-03-18:
I have this bug sometimes too, when using the "show traffic" option.

Below are Windows Error Report logs. The problem appears to be a low-level segmentation faut-type of error, e.g. maybe Access Violation Exception. Exception code 0xc0000374 is "Heap Corruption", so maybe a buffer overrun as daxxx said.

````
Faulting application name: omodscan.exe, version: 1.7.0.0, time stamp: 0x65f146c4
Faulting module name: ntdll.dll, version: 10.0.19041.5438, time stamp: 0xab0dece3
Exception code: 0xc0000374
Fault offset: 0x00000000000ff3c9
Faulting process id: 0x3924
Faulting application start time: 0x01db96cc915ef272
Faulting application path: C:\Program Files\Open ModScan\omodscan.exe
Faulting module path: C:\Windows\SYSTEM32\ntdll.dll
Report Id: 14f8c7ec-c2ca-40f5-88b9-f4403fa8cae2
Faulting package full name: 
Faulting package-relative application ID: 
````

````
Fault bucket 2244770124576194464, type 4
Event Name: APPCRASH
Response: Not available
Cab Id: 0

Problem signature:
P1: omodscan.exe
P2: 1.7.0.0
P3: 65f146c4
P4: StackHash_2d13
P5: 10.0.19041.5438
P6: ab0dece3
P7: c0000374
P8: PCH_81_FROM_ntdll+0x000000000009E044
P9: 
P10: 

Attached files:
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREEA1.tmp.dmp
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREF3F.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREF5F.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREF5F.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREF7F.tmp.txt

These files may be available here:
\\?\C:\ProgramData\Microsoft\Windows\WER\ReportArchive\AppCrash_omodscan.exe_3e55a677899395b62ef8e149bba9903295e65ff_4d2d5258_f85a443e-6419-403d-8d7b-fb49053f9a68

Analysis symbol: 
Rechecking for solution: 0
Report Id: 14f8c7ec-c2ca-40f5-88b9-f4403fa8cae2
Report Status: 268435456
Hashed bucket: fe03f6be10a698041f270688f5c5ffa0
Cab Guid: 0
````

````
Fault bucket 2227765310916770155, type 5
Event Name: FaultTolerantHeap
Response: Not available
Cab Id: 0

Problem signature:
P1: omodscan.exe
P2: 1.7.0.0
P3: 65F146C4
P4: ffffbaad
P5: 
P6: 
P7: 
P8: 
P9: 
P10: 

Attached files:
\\?\C:\Windows\SERVIC~1\LOCALS~1\AppData\Local\Temp\FTHEE05.tmp\fthempty.txt
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREE25.tmp.WERInternalMetadata.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREE55.tmp.xml
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREE53.tmp.csv
\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WEREE73.tmp.txt

These files may be available here:

Analysis symbol: 
Rechecking for solution: 0
Report Id: 4da563a2-5f78-4bad-ab90-d3fa1ae9985f
Report Status: 268435456
Hashed bucket: 74276b369c09e7cedeea9cbf4021b96b
Cab Guid: 0
````