sanny32/OpenModScan #34: Write register window edit wrong address

ghostolo90, 2025-03-05:
When I try to write a register, the new value is set on the address-1.

Same behaviour on both coil status and holding register.

Steps to reproduce:

- Double click on a value to write or its address
[image]

- Edit the value and press OK
[image]

- The value edited is set on the address -1
[image]

sanny32, 2025-03-05:
Please test at version [1.9.0](https://github.com/sanny32/OpenModScan/releases/tag/1.9.0)

ghostolo90, 2025-03-05:
Done, not working.

sanny32, 2025-03-05:
So, can`t reproduce.
Can you add video that reproduce bug?

ghostolo90, 2025-03-06:
Yes, i'm using a Beckhoff PLC

[image]

sanny32, 2025-03-07:
Please follow the steps below
1. Connect to your modbus device
2. Open Display Definition dialog (menu Setup ->Data Definition)  and set Scan Rate to 10000
3. Through the menu Setup -> Text Capture make a log.txt file

[image]

5. Make a write to registers
6. Through the menu Setup -> Capture Off stop a log capture
7. Upload log.txt file here

ghostolo90, 2025-03-10:
Hello, 
Done, log and video attached.

[log.txt](https://github.com/user-attachments/files/19162285/log.txt)

https://github.com/user-attachments/assets/b3647702-e79e-4da1-aa9f-885f4c995f52

sanny32, 2025-03-10:
Hello @ghostolo90. Thanks for what you did to understand how the bug occurred.
Based on the log data I see that you send WRITE SINGLE REG request to address 30 (0-based).

[image]

Then, application send READ HOLDING REGS request to read 31 holding register data starts at address 0 (0-based)

[image]

The response from the server returns the value at 29 address (0-based)

[image]

I suppose that problem at the modbus server.