sanny32/OpenModScan #39: Build error

GillesBA, 2025-04-02:
~/omodscan/OpenModScan/omodscan$ make
g++ -c -pipe -O2 -std=gnu++1z -Wall -W -D_REENTRANT -fPIC -DAPP_NAME="\"Open ModScan\"" -DAPP_DESCRIPTION="\"An Open Source Modbus Master (Client) Utility\"" -DAPP_VERSION=\"1.9.0\" -DQT_NO_DEBUG -DQT_PRINTSUPPORT_LIB -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_SERIALBUS_LIB -DQT_SERIALPORT_LIB -DQT_CORE_LIB -I. -Icontrols -Idialogs -Imodbusmessages -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtPrintSupport -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialBus -isystem /usr/include/x86_64-linux-gnu/qt5/QtSerialPort -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I. -I. -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -o functioncodecombobox.o controls/functioncodecombobox.cpp
controls/functioncodecombobox.cpp: In member function ‘void FunctionCodeComboBox::setInputMode(FunctionCodeComboBox::InputMode)’:
controls/functioncodecombobox.cpp:87:96: error: call of overloaded ‘arg(QString, ModbusFunction)’ is ambiguous
   87 |         setItemText(i, QString("%1: %2").arg(formatFuncCode(funcCode), ModbusFunction(funcCode)));
      |                                                                                                ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:47,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qiodevice.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qdatastream.h:44,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qregion.h:49,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/QKeyEvent:1,
                 from controls/functioncodecombobox.cpp:1:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:298:31: note: candidate: ‘QString QString::arg(const QString&, int, QChar) const’
  298 |     Q_REQUIRED_RESULT QString arg(const QString &a, int fieldWidth = 0,
      |                               ^~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:301:31: note: candidate: ‘QString QString::arg(QStringView, int, QChar) const’
  301 |     Q_REQUIRED_RESULT QString arg(QStringView a, int fieldWidth = 0,
      |                               ^~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:987:16: note: candidate: ‘QString QString::arg(const QString&, const QString&) const’
  987 | inline QString QString::arg(const QString &a1, const QString &a2) const
      |                ^~~~~~~
controls/functioncodecombobox.cpp: In member function ‘void FunctionCodeComboBox::addItem(QModbusPdu::FunctionCode)’:
controls/functioncodecombobox.cpp:102:76: error: call of overloaded ‘arg(const QString&, ModbusFunction)’ is ambiguous
  102 |     QComboBox::addItem(QString("%1: %2").arg(code, ModbusFunction(funcCode)), funcCode);
      |                                                                            ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:47,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qiodevice.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qdatastream.h:44,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qregion.h:49,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qevent.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/QKeyEvent:1,
                 from controls/functioncodecombobox.cpp:1:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:298:31: note: candidate: ‘QString QString::arg(const QString&, int, QChar) const’
  298 |     Q_REQUIRED_RESULT QString arg(const QString &a, int fieldWidth = 0,
      |                               ^~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:301:31: note: candidate: ‘QString QString::arg(QStringView, int, QChar) const’
  301 |     Q_REQUIRED_RESULT QString arg(QStringView a, int fieldWidth = 0,
      |                               ^~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qstring.h:987:16: note: candidate: ‘QString QString::arg(const QString&, const QString&) const’
  987 | inline QString QString::arg(const QString &a1, const QString &a2) const
      |                ^~~~~~~
make: *** [Makefile:1545: functioncodecombobox.o] Error 1

sanny32, 2025-04-02:
Which qt version used?

GillesBA, 2025-04-02:
@sanny32 
omodscan/OpenModScan/omodscan$ qmake --version
QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu

GillesBA, 2025-04-02:
Oh yes my mistake, thank you

closed 2025-04-02 by sanny32