sanny32/OpenModScan #41: Coil status and input status incorrect value representation

stianbauge, 2025-04-09:
Great application, however the last version 1.9.0 has an error when representing coil status and input status in the OpenModScan. Instead of the expected digital values "0" and "1" it shows "0" and "256". I'm using the prebuilt Windows versions. When I roll back to previuos version 1.8.2 this is not an issue, so it seems to have appeared int the latest version.

[image]

sanny32, 2025-04-09:
Hello @stianbauge.
Can you write modbus log (via menu Setup->Text Capture) and upload it here?

stianbauge, 2025-04-09:
Sure, here is the log file.

[coils_status_capture.txt](https://github.com/user-attachments/files/19665318/coils_status_capture.txt)

sanny32, 2025-04-09:
Thank you for your interest in the program.
At the moment I see that everything is fine in the logs.
I have not yet been able to reproduce the problem.

sanny32, 2025-04-10:
I can reproduce problem and fix it at next release.

commit 49078ccf (2025-04-10): fix #41