sanny32/OpenModScan #44: 99-omodscan.rules

Chadlucc, 2025-05-15:
i changed 99-omodscan.rules but it do not work

sanny32, 2025-05-15:
Which changes did you made?

Chadlucc, 2025-05-15:
> Which changes did you made?
 
such as ：  KERNEL=="ttySSSS[0-9]*", GROUP="dialout", MODE="0666"

[image]

ttyS1 ttyS2 .....  still list

sanny32, 2025-05-15:
KERNEL=="ttySSSS[0-9]*", GROUP="dialout", MODE="0666"
I think there are extra S's here (ttyS**SSS**[0-9])

Chadlucc, 2025-05-15:
> KERNEL=="ttyS**SSS**[0-9]*", GROUP="dialout", MODE="0666" I think there are extra S's here

yes,  there are extra S's here, but ttyS1, ttyS2, shoud Not meeting this rule,sp it should not display in the UI

Chadlucc, 2025-05-16:
> The file 99-omodscan.rules does not affect the display of the list of devices, but only access to them under a regular user.

if i want UI display ttyRS485-1, ttyRS485-2,........     How should i do?

sanny32, 2025-05-16:
Have you ttyRS485-1, ttyRS485-2, ... devices? Please, print `ls /dev/ttyS*`

Chadlucc, 2025-05-16:
> ls /dev/ttyS*

[image]

What I mean is whether relevant configurations can be made to allow users to decide which devices to display themselves

sanny32, 2025-05-21:
I don't know how to solve this problem yet. More research is needed.

closed 2025-06-24 by sanny32